AlexanderKoch-Koch/low_cost_robot #5: Unclear if wrist is supposed to be rotatable during assembly

moaazsidat, 2024-04-03:
I'm trying to figure if the wrist is supposed to be rotatable during the assembly since I didn't quite catch that from the youtube video.

Are the screws here correct in being at the top and bottom, or do they need to be left and right? 

I'm unable to rotate the wrist joint during assembly whereas all the other ones are rotatable by hand, is that expected?

[image]

AlexanderKoch-Koch, 2024-04-04:
I think the screws you used might be too long and block the rotation. You have to use the shortest screws here. You can just put in all four screws. I put in only two since it's faster and strong enough usually.

moaazsidat, 2024-04-04:
Ok, that was probably it, used smaller screws and applied a bit more force on manual rotation and that seemed to get it goin. It still seems a little frictiony on the rotation, so I might have gotten a bad servo.

closed 2024-04-04 by AlexanderKoch-Koch