AlexanderKoch-Koch/low_cost_robot #6: Longer servo connector

trzy, 2024-04-06:
Hi Alexander, thank you for open sourcing this. I printed out the components and noticed that there is a longer connector piece that would replace the component labeled "servo connector angle v1:1". I printed it too dense, so I won't use it, but I am going to print a lighter version of it. I see an even longer version in the photograph in the README.md. Is there any major disadvantage to using these longer versions? I assume you shortened it for stability purposes?

AlexanderKoch-Koch, 2024-04-06:
It would not replace the "servo connector angle". You could use both to get a 5dof arm. The length of it does not really matter. A longer piece gives you a larger reach but is more difficult to control. I think the version in this repo has a good size.

trzy, 2024-04-07:
I see now looking more closely at the videos. This makes sense! Thank you. I've gotten the 4dof version built and working!

closed 2024-04-07 by trzy