AlexanderKoch-Koch/low_cost_robot #7: How do you wire the leader arm + where does the 6th servo go?

ControlCplusControlV, 2024-04-06:
[image]
I've assembled the leader arm, but while the README calls for 6 servo motors I only ended up using 5? And I think I assembled it correctly? Also how does the wiring work for the leader arm, not sure how to do that, I'm guessing loosely like the follower but idk specifics.

Thanks for open-sourcing though! Looks sick and can't wait to try it out

ControlCplusControlV, 2024-04-06:
[image]
So it seems like that `extra` arm piece goes in the middle of that top segment? I don't think the images in the README show that, although it is in the stl files

AlexanderKoch-Koch, 2024-04-06:
You can use it with either five or six servos. The elbow_to_wrist_extension is the additional part. I designed it originally with only five servos. That's why the readme renders show only five servos.

ControlCplusControlV, 2024-04-06:
Ah I see, thanks! I did manage to get it working with the elbow_to_wrist_extension just now. Do you have a wiring diagram or more clear pictures on how to wire the leader arm? Not sure how I should set it up

AlexanderKoch-Koch, 2024-04-06:
The wiring for the leader arm is simpler. You can directly connect the servo control board to the first servo motor and then daisy chain all the others since all servos in the leader arm use 5v.

ControlCplusControlV, 2024-04-06:
Oh I see, works great, thanks!

closed 2024-04-06 by ControlCplusControlV