AlexanderKoch-Koch/low_cost_robot #14: Any ideas on how to debug flakey dynamixel connections

moaazsidat, 2024-04-11:
Having trouble consistently being able to connect to motors and scan them via dynamixel, wondering if there's ways to debug it more

* When attaching them one by one, it works _most_ of the times, and nothing shows up on scanning via Dynamixel wizard rest of the time
* When trying to connect one motor to another, nothing shows up on scanning via Dynamixel wizard 100% of the time
* When have the whole setup wired, the smaller motors are blinking red (seems like that's incorrect voltage error) – and nothing shows up on Dynamixel wizard scanning

AlexanderKoch-Koch, 2024-04-11:
The Dynamixel wizard shows you the input voltage. You could check that this is correct. Otherwise, I have no idea.

moaazsidat, 2024-04-11:
Gotcha, will debug more, thanks!

closed 2024-04-11 by moaazsidat