AlexanderKoch-Koch/low_cost_robot #15: Might want to recommend a different way to set the voltage

johnowhitaker, 2024-04-13:
"Connect to an XL330 servo and view the input voltage on Dynamixel Wizard, then adjust the screw on the voltage reducer until the input voltage is 5V"

My voltage reducers came set such that the output was 12V and needed many turns of the pot to start bringing that down, by which time the poor little servo had let the magic smoke out. Might have been user error my side but for my second attempt I used a PC fan instead, which was happy with 12V input until I had time to set the output voltage down to 5V.

AlexanderKoch-Koch, 2024-04-13:
Interesting. I haven't been able to destroy a servo this way. Are you sure it's dead? There is a recovery feature in the Dynamixel Wizard.

johnowhitaker, 2024-04-13:
[image]
Burnt beyond recovery 😂 There's a small chance I had wires switched but I'm too chicken to try on another servo to find out. Possibly also caused by connecting the servo *after* power and USB?

karanchahal, 2025-01-13:
Yeah the instructions are pretty vague, the issue is the dynamixel motors dont clearly specify what is ground, + and -ve in their diagram.

I think we might be doing it wrong, plus the instructions have some sort of magicness to them.

Connect the D and G to the voltage reducer ? WHy, I would expect Ground to be untouched and simply attached to ground of the servo.

And the V and D (+ve and -ve) to be atatched to the voltage reducer and hooked up the + and -ve of the servo.