AlexanderKoch-Koch/low_cost_robot #16: Apple Slilicon Issues with Dynamixel Wizard

ControlCplusControlV, 2024-04-16:
Is there any way around Dynamixel wizard? I'm on M1 Apple Silicon, and the App Store installation like they suggest does not work, Dynamixel wizard either doesn't detect the motors (usb c into the servo board for the 6v motor, attempting to scan with jumper wires into the motor, tried both orientations, tried with additional dc power, and without (on the other port) and it either won't detect the motor at all, or it will detect a motor on every single id, not sure what to do, appreciate any advice and can close if its not relevant

edward-io, 2024-04-16:
I have an M1 max macbook pro, and the Dynamixel Wizard worked fine when using the Waveshare board on both dynamixels servos when powered with 6v DC. Are you identifying the motors one at a time?

LouisFoucard, 2024-04-23:
Did you figure out the issue? I am also having trouble with the Dynamixel wizard, on Linux. Installed Dynamixel_sdk, followed the steps in https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#basic-features to install the wizard, but the scan does not find any connected motors. I am powering the board with 12v, and I see the XL430 blink once red, so seems to be powered correctly, and I can see the board connected at '/dev/ttyACM0'.
I also tried running the `ping.py` from dynamixel sdk, its able to open the port and change the baudrate, but getting no status packet from the XL430.
```
Succeeded to open the port
Succeeded to change the baudrate
[TxRxResult] There is no status packet!
```

LouisFoucard, 2024-04-23:
started a thread in robotis' forum to https://forum.robotis.com/t/dynamixel-not-seeing-any-device/5697

LouisFoucard, 2024-04-25:
Finally got it to work: it was a bad board. Ordered another waveshare controller and everything worked fine. They seem to have a pretty high rate of failure, but even if you have to buy 2 its still cheaper than the u2d2..

peeeanut, 2024-04-29:
Can you show me your wires? I have the same issue as you.

LouisFoucard, 2024-05-01:
[image]

LouisFoucard, 2024-05-01:
[image][image]

Make sure you try a couple of these waveshare boards though, I think a lot of them aren't working

beduffy, 2024-05-05:
I think I have the same problem but not sure, I'm on ubuntu and not detecting anything in dynamixel wizard but also not getting scenario of robot on all IDs. I sometimes get a faint red light on the dynamixel which soon after disappears but find it hard to reproduce this light.

Initially, I want to control just 1 dynamixel servo, therefore I won't have the yellow wire going into D in the above diagram. @LouisFoucard do you think I will be able to control just 1 XL330-M288 this way or is the yellow wire needed for powering the daisychain with waveshare? I will order a 2nd dynamixel in case this is needed. Also considering ordering another waveshare or even the U2D2.

LouisFoucard, 2024-05-06:
For just testing one of the base motors you don't need the yellow data wire (nor the other 2 power wires since the base and shoulder motors get their power directly from the wave share board).
The faint red light is expected, it's just the motor turning on. If it blinks it means the input voltage is too high.

Most likely though you have a faulty waveshare board (if nothing works after checking all wires). My board was faulty and at least one other person also had a faulty board.