AlexanderKoch-Koch/low_cost_robot #20: How do you control the robot arm?

johnnevins, 2024-04-29:
Hi, thanks much for sharing this low-cost solution and saw your demo video on X which is great. 
Wonder did you use a local PC with GPU to implement the inference of imitation learning? Would it be possible to use a remote server with GPU to control the arm? 
I've read through the readme, but not able to find how the algorithm communicates with the hardware, could you elaborate more on this? 

Appreciated
Patrick

AlexanderKoch-Koch, 2024-04-29:
I've used my macbook directly to interface with the servos. In newer demos, I've used Raspberry Pis on the robot arms to connect them via ethernet to a central server.

johnnevins, 2024-04-30:
> I've used my macbook directly to interface with the servos. In newer demos, I've used Raspberry Pis on the robot arms to connect them via ethernet to a central server.

Thanks much for the reply, as for the Raspberry Pis& server combination you mentioned, how do you translate output of the algorithm(position control I guess?) to executable code for hardware? Do you use GPIO? 

Bests:)

closed 2024-09-10 by johnnevins