AlexanderKoch-Koch/low_cost_robot #31: the 3D printed base for follower arm is not strong enough?

jimtangshfx, 2024-06-09:
Hi, 
Thanks for sharing this project. 
I Printed the base of follower arm and height of the bottom base is only around 3mm , so the one I printed out using PLA at infill 60% is very soft and weak, not strong enough. I wonder what material are you using and how come yours looks very strong? 
Thank you.
[image]

AlexanderKoch-Koch, 2024-06-09:
I'm using standard PLA. When you use the table clamp to fix it to a table it becomes quite strong since you can put the clamp very close to the first servo motor.

jimtangshfx, 2024-06-09:
> I'm using standard PLA. When you use the table clamp to fix it to a table it becomes quite strong since you can put the clamp very close to the first servo motor.

Hi, Alex
Thanks for your quick reply. I wonder why my printed base is so weak that I can even bend it using my hand?
what's your infill percentage and wall thickness? I am using below settings.
Layer height: 0.3mm
Wall thickness: 0.8 mm
Top-bottom thickness: 1.1mm
Infill: 60% Grid