AlexanderKoch-Koch/low_cost_robot #36: Waveshare board issue?

yudhisteer, 2024-06-24:
Hi,
I bought two waveshare board and when testing the motors, it is not being recognized. What is the probability that two of the waveshare board is not good? 

[image]

[image]

ex3ndr, 2024-06-29:
You need to supply power to them to power up

MichaelrMentele, 2024-07-08:
I'm not seeing the motor either.
[image]

I do see the board:
[image]

But scan finds nothing. Tried two boards. Swapped the motors.

MichaelrMentele, 2024-07-09:
> Hi, I bought two waveshare board and when testing the motors, it is not being recognized. What is the probability that two of the waveshare board is not good?

It looks like you can detect the motor when doing the firmware updates though based on your screenshot?

MichaelrMentele, 2024-07-09:
For me it's definitely the power supply. The 5v won't power the 330 for more than a moment. The 12 v causes the 330 to burn up but won't power the 430 for more than a moment.

MichaelrMentele, 2024-07-17:
@yudhisteer any update?

MichaelrMentele, 2024-08-13:
I had one bad board -- got a new one and it worked. Makes you want to just use the U2D2

karanchahal, 2025-01-13:
Yeah for sure, U2D2 would have helped I think. But now I am in too deep, I burnt 2 waveshares in this process.