AlexanderKoch-Koch/low_cost_robot #38: Waveshare connector doesn't fit the servo wire

MichaelrMentele, 2024-07-08:
The connectors do not fit together. If this is expected it would be good to extend the instruction to trim the connector.

[image]

MichaelrMentele, 2024-07-08:
Highly likely I am confused -- unfamiliar with all this. Going off of the dynamixel quickstart example. https://www.youtube.com/watch?v=JRRZW_l1V-U 

Screenshots for each step would be great.

MichaelrMentele, 2024-07-17:
I ended up clipping off the outter edge but can't seem to keep motors powered.

moaazsidat, 2024-08-02:
@MichaelrMentele you'll need male-female jumper wires to connect these

meetsitaram, 2024-08-18:
I used extra wires to connect it this way.
[image]

karanchahal, 2025-01-13:
What is +ve, -ve and GND on the white connector of the dynamixel ?

I think the D, V and G in the waveshare means.
-ve, +ve and ground