AlexanderKoch-Koch/low_cost_robot #39: Linked Power Supplies Don't Power their Respective Motors

MichaelrMentele, 2024-07-09:
When I use the 5v adapter it doesn't power the 330's for more than a brief moment on plug in. 

When I use the 12v the motor it powers the 330's but starts to burn up as @jzxchiang1 experienced in #26 

[image]

catid, 2024-07-12:
You might have the cables in the wrong places.  Use the male/female jumper wires to swap them

MichaelrMentele, 2024-07-17:
Thanks Chris but I've flipped the jumper end orientation but it doesn't make a difference. Do you have a screenshot of your working board @catid? I wonder if my waveshare board is bad or was ruined during one of the permutations.

meetsitaram, 2024-08-18:
My waveshare board got ruined, most probably because i might have connected wrong wires. When i tried scanning for servos using the dynamizel wizard, i was getting a warning on windows saying it is unable to connect to the hardware properly. Had to order a new board.

karanchahal, 2025-01-13:
the 5v adapter will not help as the dc jack can only support 9-12 v